km4ack/pi-build #3: Add LinBPQ to install script

km4ack, 2019-12-30:


VA7DGP, 2020-01-28:
I had Linbpq running several years ago. I could not get the SCS-PTC tnc to work. (I had help from a Linux guru) A few months ago. I tried to get a 9612 (Prolific adapter), KPC3+ (RT serial adapter) and a RT system 3.5 MM plug with a patch cord to run VFO. I gave up. It seem my luck with the PI (linux) is doom for failure

KI7ODK, 2020-09-12:
@VA7DGP LinBPQ works great with Direwolf. This is how I run my portable Winlink gateway on my Pi.

There is a separate install script for using linBPQ as a Winlink gateway. I don't know if it would be worth it to include that as a separate option or just install linBPQ regularly and then reconfigure it as users need.

VA7DGP, 2020-09-12:
I had Kantronic 9612 tnc, KPC3+ and SCS-PTC tnc with RT system 3.5 MM plug for CIV control. I had a previous bpq32.cfg file. I just changed the comports to /dev/ttyUSB0 etc. 

 

I could get the tnc to talk to the radio and back. 

 

I worked at for 2 Months and gave up.

 

73

 

Don va7dgp

km4ack, 2020-10-14:
slated for 3.0.5 release. Currently in dev branch.

closed 2020-10-14 by km4ack