km4ack/pi-build #10: Add RTC Module to list

nrdingle, 2020-01-28:


WheezyE, 2020-01-29:
I already put together a list of commands for this RTC module in the past. I'm not sure about making it a standard installation package on the Pi Build script though, since not everyone will have purchased this module and they may think it's a script to set up software (not a new hardware component).  It might be good to include a description that this is a hardware chip, and the script can set it up if needed/installed on the pi.  https://pastebin.com/bU9exSA3

km4ack, 2020-01-29:
Thanks @WheezyE I'll probably write a separate script that the user can run if needed instead of including it in the main script. 73, de KM4ACK

km4ack, 2020-02-04:
Script has been written. Need to test before adding to git hub. This WILL NOT be written into the build a pi script as the user must enable the I2C interface before running the script. Video coming soon 😃

WheezyE, 2020-02-05:
> Script has been written. Need to test before adding to git hub. This WILL NOT be written into the build a pi script as the user must enable the I2C interface before running the script. Video coming soon 😃

Just an update (I haven't had a chance to try this yet), but enabling I2C devices may be possible via the command line without having to use raspi-config.  Some authors recommend removing i2c-bcm2708 from /etc/modprobe.d/raspi-blacklist.conf (by placing a hash ‘#’ in front of "blacklist i2c-bcm2708")

https://ozzmaker.com/i2c/
https://www.instructables.com/id/How-to-enable-I2C-on-RaspberryPI/

km4ack, 2020-02-05:
Reopening this issue based on @WheezyE last comment. Testing needed to confirm this works with buster. If anyone has luck with this please let me know. It may be next week before I have time to dig into it.

WheezyE, 2020-02-16:
I finally had some time to look into this and figured out how to enable i2c from the command line.

Type: sudo nano /boot/config.txt
Then just un-comment "#dtparam=i2c_arm=on" so that it looks like this: 

```
# Uncomment some or all of these to enable the optional hardware interfaces
dtparam=i2c_arm=on
``` 
After exiting/saving and the file, check out the raspi-config or Desktop 'start > Preferences > Raspberry Pi Configuration > Interfaces" tab - I2C should be enabled.

km4ack, 2020-02-17:
I can use sed to uncomment the line in the script. Did you have to reboot for the change to take effect?

km4ack, 2020-02-17:
> I'm not sure how to uncomment a line via a script though.

I can use sed in the script to make this change

km4ack, 2020-02-21:
added to dev branch for testing

km4ack, 2020-02-21:
tested and added to master branch. still requires reboot after mod of /boot/config.txt so option is given on first install screen and if checked mods the /boot/config.txt file. After first round of software is installed and pi reboots, the rtc script is downloaded and ran to complete the installation.

closed 2020-02-21 by km4ack