km4ack/pi-build #11: Add WSJT-X log file viewer

LY2BHP, 2020-01-28:
It would be just great to have a simple WSJT-X log file viewer as a separate small app.

km4ack, 2020-02-21:
Does an app like this already exist that can be run on the pi? If so, do you have a link?

LY2BHP, 2020-02-21:
Not that I’d known of, that’s why I’d started this issue here. I only know (and use) a very nice app LoggerWS by  LY2CG but it is for Windows only. You may download it here ( http://qrz.lt/ly2cg/) just for better imagination what am I looking for for Raspberry Pi.

km4ack, 2020-02-21:
ok. I'll check out the link but will close this issue. This would be a new project if it is something I decide to take on in the future. The build-a-pi script is just a way to install existing apps on the pi to help operators get up and running quicker. Thanks for the link.

closed 2020-02-21 by km4ack

LY2BHP, 2020-02-21:
Thanks for the build-a-pi tutorials! All were extremely helpful for me! As
your channel’s subscriber I’m looking fwd to more pi-related ham ideas
incl. the one for WSJT-X log viewing! 73! Linas LY2H/LY2BHP