km4ack/pi-build #26: No Audio Device Settings item in the Preferences  menu

LY2BHP, 2020-02-27:
The Audio Device Settings item disappeared from the Preferences menu after the full build-a-pie installation on the brand new Buster-loaded card.   It is also not on the list under the Main Menu Editor nor under any other Main Menu section. I need this Audio Device Settings item in order to configure my Digital Mode Radio Interface. All the other GUI items are in place and all ham apps of the build work just nice. I wonder if this could be a bug? If so, how to fix it? 73! Linas LY2H

km4ack, 2020-02-27:
This has nothing to do with the build a pi script. 

The audio device setting are found in the top right corner of the desktop in the speaker icon. Right click on the speaker to view your USB device. 

Another approach is to open terminal and run alsamixer

closed 2020-02-27 by km4ack

kf4mot, 2020-03-26:
This got me too, but alsamixer allowed me to fix it.
Nothing wrong with the script, but something did change somewhere because we use to be able to do this at the speaker icon. Either happened because of Pi4 upgrade or the Buster upgrade.

km4ack, 2020-03-26:
KF4MOT,
Which sound card are you using? Are you able to right click on the speaker and choose your USB sound device and the right click again on the speaker icon and choose USB Device Settings to access the settings?

kf4mot, 2020-03-26:
It's the same $4 C-media dongle that we all know and love. The complication is that because I'm using vox for ptt on the right output and data on the left output the 2 output channels need to be set separately to not overdrive the tx, yet still be able to key tx.
I can select the device and change settings at the speaker icon. I just can't set them correctly.
If I remember correctly I have one slider for the mic when I'm in the mic tab. However on the output tab it shows a speaker slider and a mic (!?) slider where the left/right out sliders should be. They're not just mislabeled because they don't function correctly either. If you try to adjust the speaker "mic" it just snaps back to its default setting.

I could get you a pic tonight if that's not clear and you're still curious.

km4ack, 2020-03-26:
Thanks for the explanation!