km4ack/pi-build #27: Direwolf Flashing

Tandil1368, 2020-02-27:
I have installed Buster on a Pi3B+ and run the script twice with the same issue.  When I run Direwolf through the menu I get a connection refused port 4525 error.  If I run Direwolf manually or through the GUI it flashes constantly.  I am using it with FLRIG and an ICOM IC-9100 direct to the Pi.  The 9100 has a built-in sound card and everything is working except the constant flashing on Direwolf.  I have even tried the switch to prerelease version as was suggested elsewhere to no avail. 

Also , which config file can I manually put in my Grid, Long and Lat without using an external GPS so the information will show properly on the right side menu.

I want to thank you Jason for this fantastic setup and all your hard work.

km4ack, 2020-02-27:
Flashing direwolf is a known issue on Buster. This is up to the direwolf team to resolve. The workaround is to start direwolf with 'direwolf -p -t 0' (Note: that is a zero at the end) The port error is likely a direwolf configuration error with your particular rig or a port error between Pat Winlink and your rig for rig control. Not owning a 9100, it will be near impossible for me to diagnose. You can change the way direwolf starts and the rig control commands in the pat menu config file (~/patmenu/config)

The lat and lon are pulled from the GPS. It you wanted to enter manually, you will need to modify your .conkyrc file (Note the '.' in front of the filename). It is a hidden file in your home directory.

closed 2020-02-27 by km4ack

Tandil1368, 2020-02-27:
Thank you Jason, that worked great.  

--
73 de Chris NO7E