km4ack/pi-build #30: Headless operation

K4BQN, 2020-02-28:
Jason,

First, thank you for the hard work, your builds are outstanding. One quick note that I just noticed, it seems that with the new PI4 B you cant run headless unless you modify the Config.txt because the PI is looking for an HDMI monitor. Adding the following statement, Adding hdmi_force_hotplug=1 to /boot/config.txt  helps, but the screen definition changes once you connect thru RealVNC. I will appreciate any advice on this issue.

Thank you,

Ed

km4ack, 2020-02-28:
see this video for the fix. https://youtu.be/Uy6xbC0Hiew

closed 2020-02-28 by km4ack

K4BQN, 2020-02-28:
Thank you for the prompt response.

Ed

km4ack, 2020-02-28:
Welcome. As far as I know, this isn't something I can correct with the script since the pi is already booted when the user runs the script. The issue has to be resolved by the end user.

K4BQN, 2020-02-28:
Jason,

Thank you for the great support. I was able to play with the settings, and change some values, and looks very good now. Looking forward to watch the upcoming videos.

Once more thank you, and have a great weekend.

Ed