km4ack/pi-build #31: Pi-Build v2 unexpected exit

N1JCM, 2020-02-29:
Hello Jason & Group:

After video, I ram the Pi-Build v2 script on my Pi4 (basic buster load). I seemed to get through the 1st list-based build and re-booted, but then put up a dialog about an error and also the 2nd list. When I tried to copy the text of the error, the screen cleared of both boxes and now I have a startup screen and no apparent way to get the 2nd list apps installed !?! On my desktop is just the Pat Menu icon and Conky. Any thoughts on what happened and how to recover without starting from scratch?

Thanks...

km4ack, 2020-02-29:
If the second half of the script didn't complete, a reboot should show you the second inbox again. This initial issue could have been caused if you selected the RTC but didn't actually have a real time clock installed on the pi.

N1JCM, 2020-02-29:
Thanks - I did not select the RTC (but I did select pulse, which was un-selected upon open of list). Here is the message I got in an "add-on build-a-pi" terminal box:

(yad:925): dbind-WARNING **: 21:58:43.780: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files. 

Any help?

N1JCM, 2020-02-29:
FYI, a reboot did bring up the two boxes again, including the 2nd list...

km4ack, 2020-02-29:
Interesting. That is a YAD warning. Yad is the software that displays the box that allows you to select the software. I'm guessing just a glitch since no one else has reported this behavior. Since the second box came up after reboot, you should be ok to continue with the install.

N1JCM, 2020-02-29:
Will-do and I'll report back when completed (or crashes :-)

N1JCM, 2020-02-29:
Another FYI - if I close the addon-build-a-pi terminal box, the 2nd list box is closed as well, so I'll leave it open!

N1JCM, 2020-02-29:
in re: above FYI ===> DUH!

closed 2020-03-04 by km4ack