km4ack/pi-build #33: Best practice for executables

JaySheepdog-N0SR, 2020-03-04:
Instead of using ~/bin for executable files & scripts, should consider placing them in 
/usr/local/bin
which is the standard directory for added on program executable files. You will notice that when building the FLdigi suite from source that this is where the files reside.
This will eliminate issues for those of us who do not use the “pi” user on our RasPi but may have run the script as the pi user initially.

Swift34, 2020-03-08:
I would agree that /usr/local/bin should be the place for all locally built binaries and would suggest one more thing...unless you are debugging the binary executables and shared libraries (using a debugger like gdb for instance) there is no reason to have the debugging information. To strip the binaries you would type:
sudo strip fldigi
As an example after compiling and installing fldigi the main executable is 85.767.936 bytes or 85.8 MB and after it is 6197000 bytes or 6.2 MB in size.
This won't change the amount of RAM that is used during runtime, but it does save on sd card space! This also works on shared libraries.
To get all the files in a directory like /usr/local/bin you could:
cd /usr/local/bin
sudo strip *
You can always examine the file types and whether or not they've been stripped using:
cd /usr/local/bin
file *

73 de KN4FTT

WheezyE, 2020-04-10:
I was looking this up recently and it appears that maybe /opt/ would be the best directory to install binaries that also have folders and files included with them.
https://askubuntu.com/questions/6897/where-to-install-programs

JaySheepdog-N0SR, 2020-04-10:
The FHS that is referred to is deprecated (written in 2003) and the discussion referred to is from 2010.

That being said, there are a few software developers who still install into /opt but the majority use the /use/local hierarchy. Binary files go into /use/local/bin and supporting files go to /use/local/share/<appname>

If you examine these directories, you’ll see that the fldigi suite of files go here as well as Xastir and many others.

Consistency is key, from an administrator’s standpoint, so if you install into /opt you should install everything into /opt. The best installation scripts give the option as to where to put the installed files for a given package, but this is not the norm.

I still contend, from my years as a Unix administrator and current industry practices, that /use/local is the most appropriate location for application installs.
73
N0SR