km4ack/pi-build #35: Add GPS support to direwolf

km4ack, 2020-03-06:


m0nde, 2020-03-06:
Step 5 in this document https://github.com/wb2osz/direwolf/blob/master/doc/Raspberry-Pi-APRS-Tracker.pdf

I ran this

cd home/pi/direwolf
make clean
then
make
At the end of the make a hint appears to install - then you get GPSD support confirmed Direwolf

km4ack, 2020-03-07:
Most of the items that need to be installed according to the referenced document are already installed with the GPS section of build a pi. It appears that the reason direwolf doesn't include GPS support now is that a reboot doesn't take place between the GPS install and the direwolf build. It may be as simple as moving direwolf to the second application install section so that direwolf is built after a reboot. This will need to be tested. We may also need to install libgps-dev but M0NDE didn't have to do this before he rebuilt direwolf.

km4ack, 2020-08-17:
At this time this issue will not be fixed in Build a Pi. It only applies to a very limited number of users who can follow the directions submitted by m0nde if this feature is needed. Documentation label added.

closed 2020-08-17 by km4ack