km4ack/pi-build #36: SDR Rsp1a

catalinalb, 2020-03-07:
Hello Sir,

Adding the SDR-Play RSP1A would be a huge step forward since it's becoming more popular considering its prices /performance.

Sincerely, 
Catalin Alb

km4ack, 2020-03-07:
The SDR-Play RSP1A is a piece of hardware. Do you have a specific app that you would like included in the build-a-pi?

catalinalb, 2020-03-19:
Hello,
So I have run the script and install it on the official raspberry image from SDR Play. Worked great.
I also have bought a USB sound card. Please help me to pipe the output from gqrx into WSJT-X. 
In windows, I would have used VB-Audio Cable. 
Is there a way to run the UDP button in gqrx and recieve the audio stream over UDP in WJTX?
Or a way to pipe the audio output to the mic input?

Thank you,
Catalin

km4ack, 2020-03-27:
Catalin, this is beyond the scope of this project. The build-a-pi script is intended to make it easy for the operator to install apps on the pi. Configuration is left up to the operator. 

this issue has been replaced with https://github.com/km4ack/pi-build/issues/49 which is install of actual SDR software.

closed 2020-03-27 by km4ack

catalinalb, 2020-03-27:
Yes,  I did it the other way around. I have runned the script directly into the raspian image from sdrplay.co.uk and it worked like a charm. Now I run gqrx with RSP1A and wsjt-x through Virtual_Sink (audio).

Thanks,
Catalin

Closed #36<https://eur02.safelinks.protection.outlook.com/?url=https%3A%2F%2Fgithub.com%2Fkm4ack%2Fpi-build%2Fissues%2F36&data=02%7C01%7C%7Cf6e190cc551441aecd2808d7d271ce48%7C84df9e7fe9f640afb435aaaaaaaaaaaa%7C1%7C0%7C637209258533168870&sdata=14BB9U8iCfEV0bPAOgoRkDtFF6Ryhog%2F9%2BHSs6A69Sk%3D&reserved=0>.