km4ack/pi-build #37: ardop not working

PD7RON, 2020-03-12:
Hello Jason
I have installed a new buster on a raspberry pi 3 B+ and then everything for the pi build V2.
At the first connection in the ardop mode I get a message you can see on the picture.
In the next connection the frequency changes but there is nothing happening and it changes back to situation at the start. (picture)
The ax25 on 2m is working without problems.
In the buster version where I have installed only the "Pat menu" the ardop mode is working without any problems.
I have compared both images to see if there is a difference but they are similar to each other as far as  i can see.
I hope someone can help me to make it work.

[image]

73 Ron
PD7RON

km4ack, 2020-03-12:
Are you using ARDOP over 2M?

km4ack, 2020-03-12:
My bad, I misread the text in the photo the first time. Which rig are you using? Which sound card are you using? Are you using CAT for PTT control or VOX?

PD7RON, 2020-03-12:
No I am using Ardop on hf.
Ron

PD7RON, 2020-03-13:
I am using a IC7100 and cat.
This configuration is working in the first image with only pat and the pat menu.
When I start ardop with 2 in the menu it changes de mode to usb-d and I see the frequency where the rig is in.
Then the first connection gives an error and the second connection to hb9ak changes the frequency to 7050 mhz it is then waiting and after a few seconds it changes back to the starting position. Without cat making to send the ic7100.

73 Ron

km4ack, 2020-03-13:
I think I see the problem now. In your pat configure file, it looks like you entered a really long grid square. You need to shorten that to 6 characters. Use the pat menu, choose option 1, and then option 1 again. Re enter your Winlink credentials and set the grid square to 6 characters. Then try the ARDOP connection again and see if the problem clears.

PD7RON, 2020-03-13:
Yes Jason this was the problem.
It is working now.
In my situation, working with the ic7100, i had to make a change in the “pat configure”.
Under ardop i changed ptt_ctrl from false to true. 
True makes the ic7100 transmitting. False there is only a frequency change but no transmit.
Jason tnx for solving this problem. It works now very fine. I like the software so as it  is made with the amateur features so together and the working of it.

73 Ron
PD7RON

km4ack, 2020-03-13:
Glad it's working Ron. 73

closed 2020-03-13 by km4ack