km4ack/pi-build #38: USB CAT Interface not usable after start

DK4FT, 2020-03-12:
Installed RPI with this script and is running fine except the USB connection to the RIG.
This interface is recognized at /dev/ttyACM0, but could not be used in FLRig.

After reconnecting the USB cable the interface is detected and bound to /dev/ttyACM and working fine and usable with flrig.

In the logs the USB device is detected and I could not find a difference between both attemts after the restart:

Bad:

> Mar 12 19:36:10 raspberrypi kernel: [    1.528838] usb 1-1.4: new full-speed USB device number 4 using xhci_hcd
Mar 12 19:36:10 raspberrypi kernel: [    1.688941] usb 1-1.4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
Mar 12 19:36:10 raspberrypi kernel: [    1.688956] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 12 19:36:10 raspberrypi kernel: [    1.688968] usb 1-1.4: Product: STM32 Virtual ComPort in FS Mode
Mar 12 19:36:10 raspberrypi kernel: [    1.688980] usb 1-1.4: Manufacturer: STMicroelectronics
Mar 12 19:36:10 raspberrypi kernel: [    1.688991] usb 1-1.4: SerialNumber: 315635703036

> Mar 12 19:36:10 raspberrypi kernel: [    4.763401] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device

Good:

> Mar 12 19:53:48 raspberrypi kernel: [ 1051.464342] usb 1-1.4: USB disconnect, device number 4
Mar 12 19:53:48 raspberrypi kernel: [ 1051.464999] pps pps0: removed
Mar 12 19:53:53 raspberrypi kernel: [ 1056.623578] usb 1-1.4: new full-speed USB device number 7 using xhci_hcd
Mar 12 19:53:53 raspberrypi kernel: [ 1056.763599] usb 1-1.4: New USB device found, idVendor=0483, idProduct=5740, bcdDevice= 2.00
Mar 12 19:53:53 raspberrypi kernel: [ 1056.763606] usb 1-1.4: New USB device strings: Mfr=1, Product=2, SerialNumber=3
Mar 12 19:53:53 raspberrypi kernel: [ 1056.763611] usb 1-1.4: Product: STM32 Virtual ComPort in FS Mode
Mar 12 19:53:53 raspberrypi kernel: [ 1056.763617] usb 1-1.4: Manufacturer: STMicroelectronics
Mar 12 19:53:53 raspberrypi kernel: [ 1056.763621] usb 1-1.4: SerialNumber: 315635703036
Mar 12 19:53:53 raspberrypi kernel: [ 1056.766537] cdc_acm 1-1.4:1.0: ttyACM0: USB ACM device

km4ack, 2020-03-12:
Which radio are you using?

DK4FT, 2020-03-12:
Its a Minion Mini (QRPver), some kind of exotic, but nice and small...
[https://qrpver.com/transceivers/all-band-10-band-hf-sdr-transceiver-minion-sdr.html](url)

worked fine with same RPi with and an amRRon installation before...

[image]

73 de Alex, DK4FT

km4ack, 2020-03-12:
Nice looking rig! Is the problem repeatable? If you reboot with the radio connected to the pi, does the pi see the radio? Does it see the radio if you remove the USB and plug it back in once the pi is booted?

DK4FT, 2020-03-13:
Yes - is repeatable:

1. After Restart the interfaces are created -> not usable
2. Disconnect/Reconnect the USB Cable the interfaces created again -> usable
3. clean shutdown of RPi and new start with connected radio -> not usable

Precondition: In all cases the rig was already powered and ready for use bootet.

km4ack, 2020-03-13:
Thanks for the info. I am not certain this pertains to something the script
did or didn't do. Without owning that rig it is impossible for me to know
for sure. It would be interesting to know if this happened on a stock build
of the latest Buster with only FLRIG installed. It seems to me that this is
more of a driver issue than anything else. Or this is related to the kernel
itself. Again, the only way to confirm this is to do a manual build with
only the FLRIG software installed. Unfortunately, that is about the only
help/advice I can give at the moment.

DK4FT, 2020-03-13:
I am preparing a 2nd sd-card and follow your suggestions.
mni tnx...

km4ack, 2020-03-13:
Please let me know if you have the same issue with the new build. 73, de
KM4ACK

DK4FT, 2020-03-14:
**Preconditions:**
- clean install of Buster with FLDigi/FLRig -> CAT = OK
- After each installation Step I forced a restart of the Pi

- autoHotspotN 
  -> CAT = OK

- Installed GPS for time sync
   -> CAT = OK

- Install conky step by step (but first just the conky without needed parts for Maidenhead Loc)
   - install conky
   - download config 
   - crontab -e
   - @reboot sleep 20 && export DISPLAY=:0 && /usr/bin/conky
   -> CAT = OK

- sudo apt-get install -y ruby2.3 
    -> CAT = OK
- sudo gem install gpsd_client 
    -> CAT = OK

- sudo gem install maidenhead  
    -> CAT = NOK

**start finding the root cause**
- increased startup timer from 20 to 60sec -> CAT = NOK
- commented call of script to get Maidenhead  Loc out in config -> CAT = OK

> #${font Arial:bold:size=18}${color White}Gridsquare ${alignr}${color Yellow}${execi 25 $HOME/bin/conky/grid | cut -c1-8}

So it seems it has something to do with the GPS client, which is started in this /bin/conky/grid script.

Thanks a lot for all you valuable videos and scripts.
For me this workaround is fine, as I don't need the Grid information in conky for ham radio operations
Don't waste any time on further troubleshooting of a single appearance of this issue.  ;-)

vy 73 de
Alex, DK4FT

km4ack, 2020-03-14:
Alex, 

Thanks for the attention to detail and the thorough trouble shooting. I would have never been able to run the issue down without the radio in front of me. With your troubleshooting skills, I can pinpoint the issue.  

sudo gem install maidenhead
-> CAT = NOK

This is a ruby gem that is installed during the pi-build when we ask for the M0IAX tools to be installed. The gem is then used in two different places. First as you saw, it is called in Conky. Second, it is used in the M0IAX tools to convert lat/long to a maidenhead gridsquare.

With this information, I would proceed as follows.

1. Conky can be installed but, as you saw, you will need to comment out the section for the grid. If you don't comment this out, it may cause Conky to not start at all.

2. M0IAX tools should probably be avoided. It relies on the maidenhead gem to get the gridsquare. While this would only be run when you clicked on "update grid" in the app, it would likely cause issues down the road. This tool is only used to aid in sending special format APRS messages through JS8Call. This can be accomplished without M0IAX tools provided the operator understand how to create the proper syntax in JS8Call.

3. GPS can still be installed for accurate time when in the field. You will only be missing the gridsquare locator on the desktop with Conky and the ability to update the gridsquare in JS8Call automatically using the M0IAX tools.

Hopefully this give guidance to anyone operating the Minion Mini radio. Thanks again for the detailed troubleshooting! 

73, de KM4ACK

closed 2020-03-14 by km4ack