km4ack/pi-build #40: Check for location of GPS in script

AF1E, 2020-03-14:
My GPS is located at "ttyUSB0" not the default of the script.  Could the script check for the location?  ls /dev/*USB* and use that to install the GPS.  My GPS is a GlobalSat BU-353-S4 USB

tzneal, 2020-05-13:
Script to detect GPS and write it to stdout. Consider this public domain:

```
#!/usr/bin/env python3
import time
import os
import termios

tty_devices = [f for f in os.listdir("/dev/") if f.startswith("tty")]

def looks_like_gps(fd):
    buf = os.read(fd, 128)
    if buf.find(b'$GP') != -1:
        return True

for tty in tty_devices:
    fn = F"/dev/{tty}"
    try:
        fd = os.open(fn, os.O_RDONLY | os.O_NONBLOCK)
        for i in range(3):
            try:
                if looks_like_gps(fd):
                    os.close(fd)
                    print(fn)
                    exit(0)
            except BlockingIOError:
                time.sleep(0.5)

        # try setting it 4800 baud and reading again
        (iflag, oflag, cflag, lflag, ispeed, ospeed, cc) = termios.tcgetattr(fd)
        ispeed = ospeed = termios.B4800
        termios.tcsetattr(fd, termios.TCSANOW, [
            iflag, oflag, cflag, lflag, ispeed, ospeed, cc])

        for i in range(3):
            try:
                if looks_like_gps(fd):
                    os.close(fd)
                    print(fn)
                    exit(0)
            except BlockingIOError:
                time.sleep(0.5)
        os.close(fd)
    except (PermissionError, OSError):
        pass

```

pleasantone, 2020-06-29:
A much better solution would be to use /dev/serial/by-id/<id of gps> so it shows up no matter what other USB serial devices are plugged in.
An alternate choice, more work, but nicer, would be to use udev to recognize all GPS class serial devices
and have it create persistent mapping to /dev/ttyGPSx. This article explains how udev works.
http://hintshop.ludvig.co.nz/show/persistent-names-usb-serial-devices/

km4ack, 2020-07-07:
Slated for 3.0.3 release using /dev/serial/by-id/ approach

pleasantone, 2020-07-07:
Also further investigation shows udev makes /dev/gps* entries.

W9CVO, 2020-08-05:
Just tried the beta and the GPS select window in installation comes up with an empty pulldown.  My GPS is through GPIO and usually comes up as /dev/serial0 or /dev/serial1 Would it would be possible to show those (and others) in the list by default as a fallback?

closed 2020-08-17 by km4ack