km4ack/pi-build #43: WSJT-X - Error in Sound Output

catalinalb, 2020-03-16:
Hello,

I know that for some reason the raspberry has a problematic sound-card... something has to be done with "alsa....?" 

How can I fix that problem and use the internal sound-card of the raspberry ? (pi3 B+)

Thank you,
Catalin

km4ack, 2020-03-16:
The internal sound card is receive only. You won't be able to use it for TX. If you are good with rx only, choose the sound card in which application you want to use. The other option is to pick up a $7 usb sound card and use it instead. It will work with both TX & RX. https://amzn.to/2waOg7S 73, de KM4ACK

closed 2020-03-16 by km4ack

catalinalb, 2020-03-16:
Yes I want to receive only since I just got my SWL callsign last week: DE1CTL :)

So please let me know exactly how to enable it. In WSJT-x  -> Settings -> Audio the Input and Output lists are empty.

I want to take this with me so that's why I need less comonents. I Want to use a SW radio or my Yaesu vx-3 for FT8 even though they don't have SSB (USB). Since there is no support for any known SDR. (RSP1a or RTL-SDR)

Thank you,
Catalin

km4ack, 2020-03-16:
Are you able to right click on the speaker icon on the desktop? Choose
audio output and select analog? Nothing in the script disables this.