km4ack/pi-build #47: Config.json empty for wl2k

alpine44, 2020-03-23:
Getting pat http server start failure due to empty config.json. Is that file supposed to be populated automatically during the setup process?

km4ack, 2020-03-23:
Did you make any changes to the installed config file or mod something with
the pat menu? Pat Winlink will create that file during installation. You
can create a new one by running 'pat http' from the command line. You will
get a callsign missing error if the file is missing but not if it is
corrupt. The missing call error is to be expected. If this is the case, you
will then need to configure the file manually by running 'pat configure'
This page will help - https://github.com/la5nta/pat/wiki

alpine44, 2020-03-23:
The problem seem to be the special characters in my winlink password that
may have to be escaped in the configuration dialog.

I did a fresh reinstall with a fake password and config.json was created.
Then changed the password manually in the file and everything works great.

Thanks for putting together this pi configuration. Very useful and nicely
done.

km4ack, 2020-03-23:
Thanks for the update. This could prove useful to someone else in the future. Glad you were able to track it down.

73, de KM4ACK

closed 2020-03-23 by km4ack