km4ack/pi-build #48: Goes to hotspot if more than 3 in wpa_supplicant

wb8erj, 2020-03-26:
Hello,

Thanks for all your work. I have 3 SSID's in my wpa_supplicant file. When I am at the 3rd location it won't recognize the wifi and goes into the hotspot mode.

When I move the 3rd wifi in the file to the first, then it connects just fine.

km4ack, 2020-03-26:
Hey Matt,

I tested this today and could not reproduce the issue here. Below is my wpa_supplicant file where only the last entry is correct. It attached first time to the entry even though it was the only the last one that was correct. 

ctrl_interface=DIR=/var/run/wpa_supplicant GROUP=netdev
update_config=1
country=US

network={
	ssid="KM4ACK-891a"
	psk="PASSWORD"
	key_mgmt=WPA-PSK
}

network={
        ssid="KM4ACK-891a"
        psk="PASSWORD"
        key_mgmt=WPA-PSK
}

network={
        ssid="KM4ACK-891"
        psk="PASSWORD"
        key_mgmt=WPA-PSK
}

73, de KM4ACK

wb8erj, 2020-03-26:
Jason thanks for taking the time to look at this.

I will look into it further on my end. As an FYI, all of the routers
involved that I am trying to connect to use the DD-WRT firmware. I am
guessing there is some sort of timing issue here. So I have changed a
couple parameters and will test more on my end.

I will keep you posted if I find anything, just in case others run into
this.

Again thank for your time on this great bit of developing and your time
into my issue.

      -- Mike WB8ERJ

wb8erj, 2020-03-27:
Jason, I think I found something. What tipped me off was when I ran your
add-wifi script. When I scanned for networks it found mine sorta.

Here is the deal, my SSID is "CIA MONITORING VAN" Your script returns just
"CIA" The problem appears to be with the spaces in the SSID, which probably
screws up a bunch of stuff!

So I changed my SSID to ItsMe and problem solved!

So I wonder if others with a space in their SSID also have this problem?

Again thanks for your work!

      -- Mike WB8ERJ

km4ack, 2020-03-27:
Thanks for the report. I have marked this issue as documentation for others here on github and added this to my FAQ https://tiny.cc/km4ack-faq under the hotspot section.

closed 2020-03-27 by km4ack

wb8erj, 2020-03-27:
One more thing, if there are any SSID's that have a space in the wpa_supplicant file it won't even connect to one that has no space. In other words, all SSID's in the file can't have a space in them.

"@type": "EmailMessage",
"@type": "ViewAction",
"target": "https://github.com/km4ack/pi-build/issues/48#issuecomment-604796079",
"url": "https://github.com/km4ack/pi-build/issues/48#issuecomment-604796079",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"@type": "Organization",
"name": "GitHub",
"url": "https://github.com"
}
}
]