km4ack/pi-build #52: Consider adding Artemis 3 (signal identification guide)

WheezyE, 2020-04-08:
I've been able to get [Artemis 3](https://aresvalley.com/Artemis/) working on the RPi3B+ and RPi4B by compiling it myself (the official Linux exe from their site doesn't work).  I can change the audio output of Artemis to the Pi's own sound port using Pulse Audio so that Artemis won't output sounds and chirps on the Sabrent USB soundcard (similar to how you showed how to control the input levels for JS8Call with PulseAudio in [your video](https://youtu.be/r0-CzyYgW5k?t=781)).

The downside to using Artemis 3 on the Pi at this time however is that - since we have to build it from scratch for the Pi - the build process takes 3 hours, takes about a gig of space, and installs Pyenv (which isn't horrible to have on the Pi, but maybe not necessary).  The resulting EXE files and created folder can be distributed though (without any need to install any other software for the end-user), but the total filesize for that package is about 170MB.  I'm in the process of addressing the Pi problem with the developers so that maybe they can host the Pi EXE in the future (like they host a Linux EXE, though that EXE doesn't work on the Pi).

[The Artemis developers (MarcoDT) have expressed an interest](https://www.rtl-sdr.com/shazam-style-automatic-signal-identification-via-the-sigidwiki-database/#comment-167727) in integrating a Shazam-style audio identification program into Artemis at some point in the future (to auto-ID audio signals heard from an SDR or receiver).  Though that "Auto ID" feature may still be a way off in development, the Artemis team also has made some progress with an "[Auto Correlation](https://aresvalley.com/documentation/)" feature which may also try to identify signals in the future by noting how often (and for how long) certain segments of signals repeat.

I'm attaching my bash instructions to compile the RPi Artemis exe here and on [PasteBin](https://pastebin.com/v3Kjf3C8) in case you want to see how it runs on your Pi:
[Build Artemis 3 2 0 from source for RPi3-4.txt](https://github.com/km4ack/pi-build/files/4452987/Build.Artemis.3.2.0.from.source.for.RPi3-4.txt)

I think that RPi3B+ would be a minimum requirement, since it takes quite a while to load (about 15 seconds) before running.

WheezyE, 2020-04-09:
Update: MarcoDT, one of the Artemis 3 developers, said he would like to host a Raspberry Pi binary of Artemis on the main page.  So it looks like the build time/space issues won't be a problem for us. 
 Artemis 3 could be included in the Build a Pi script with some simple wget and tar commands to download an unzip

WheezyE, 2020-04-10:
Another update: MarcoDT has added a Raspberry Pi download option to the main site.  Try it out on Pi with the following terminal commands:
```
cd ~/Downloads
wget https://aresvalley.com/download/1045/Artemis_3.2.0.tar
tar -xvf Artemis_3.2.0.tar
sudo chmod -R 755 artemis/
cd artemis # We must be inside the artemis folder before running Artemis or else it will say the themes folder is missing.  This isn't a problem if we run the program from a shortcut though - try running the .create_shortcut.sh script
./Artemis

```

km4ack, 2020-04-10:
Thanks for the updates @WheezyE  Currently I don't have an SDR to experiment with. I'll try to pick one up after the human malware subsides.