km4ack/pi-build #56: Cant change hot spot name or password

KN4ORP, 2020-04-16:
I can't change hot spot name or password. Here is the output i am getting.

Would you like to change the hotspot name? y/n y  
What do you want the new name to be? pi

Would you like to change the password? y/n y
What password would you like to use? pi123456
New Name=pi
New Password=pi123456
Is this correct? y/n y
Making Changes
Updating name
sed: can't read /etc/hostapd/hostapd.conf: No such file or directory
Updating password
sed: can't read /etc/hostapd/hostapd.conf: No such file or directory

km4ack, 2020-04-17:
Are you running my build-a-pi script? Did you install the hotspot?

KN4ORP, 2020-04-17:
Yes I ran your build a pi script. But I didn't name it then. When it rebooted after finishing the installation of all the software the hotspot was active with no name or password so none of my device's could see it. Is there a way to uninstall and reinstall just the hotspot?

km4ack, 2020-04-17:
I did a video on installing manually a year ago. https://youtu.be/QZtAOLcY5dY You should be able to reverse engineer but it won't be for the faint at heart. Something didn't go right with your build. You might want to try installing just the hotspot again. https://youtu.be/qMT-0mz1lkI Not sure it will work but worth a shot. Something didn't go right with your build if hostapd didn't get installed.

KN4ORP, 2020-04-17:
Ok will see what I can figure out.

jurnan, 2020-05-08:
Hi,

Thanks for the script km4ack. 

I installed it and it goes like a charm. 

However I didn’t get - as far as I am aware - the option to change the name of the WiFi network. So now I’m ‘stuck’ with RPiHotspot. I’d like to change this, but simply running the config is (unsurprisingly) not enough. Any tips? (Either through a reinstallation or config steps with the current installation. I don’t even know how to change the password in the current installation to be honest). 

Thanks!

Regards,

J

km4ack, 2020-05-08:
If you used build-a-pi, open up hotspot tools. This will allow you to change the hotspot's SSID If you didn't install hotspot tools with the build, you can still install it now https://youtu.be/O_eihSN_ES8

jurnan, 2020-05-08:
Awesome, thanks. I’ll give it a try tomorrow. 

Kudos for the speedy reply!

closed 2020-05-17 by km4ack

jurnan, 2020-05-20:
That worked. 

Hotspot works great. However, I want it permanently connected to a vpn server, to force all traffic of smart home devices through a more secure connection. I can get the vpn working, but immediately lose access to the internet when trying to connect another device. Is this port issue? Or perhaps an issue you have encountered before?

Kind regards,

Jurnan