km4ack/pi-build #58: Cloning in M0IAX Tools section fails due to attempt to log into GitHub

kcary, 2020-04-22:
Clone copy fails whether you supply an account or not.

[build-log.txt](https://github.com/km4ack/pi-build/files/4513306/build-log.txt)

[gps error.txt](https://github.com/km4ack/pi-build/files/4513308/gps.error.txt)

Thanks for this awesome BAP script!!!

km4ack, 2020-04-22:
Mark, M0IAX is currently updating his JS8CallUtilities. As soon as his latest app releases, this will be resolved. 73, de KM4ACK

closed 2020-04-22 by km4ack

km4ack, 2020-04-22:
Update....he has finished development. Sorry I missed the release. Script has been updated. You should be able to manually install with these commands:
pip3 install gps
pip3 install maidenhead
pip3 install psutil
pip3 install OSGridConverter
cd $HOME/bin
git clone https://github.com/m0iax/JS8CallUtilities_V2.git
chmod +x $HOME/bin/JS8CallUtilities_V2/gpsdGPSListener.py $HOME/bin/JS8CallUtilsGPSD/JS8CallUtils_v2.py
mv $HOME/bin/JS8CallUtilsGPSD/* $HOME/bin
rm -rf JS8CallUtilsGPSD

Now run the following:
sudo nano /usr/share/applications/js8callutils.desktop

when the file opens paste the following lines into it:
[Desktop Entry]
Name=M0IAX JS8 Tools
GenericName=M0IAX JS8 Tools
Comment=GPS & Message Tool for JS8Call
Exec=$HOME/bin/JS8CallUtils_v2.py
Terminal=false
Type=Application
Categories=Network;HamRadio;

Save and exit with ctrl+x then Y and finally ENTER

You should now have a working shortcut in your main pi menu under the hamradio section

73, de KM4ACK