km4ack/pi-build #60: GPS and Conky Issue

uksoundz, 2020-04-26:
Hi,

I have a Ublox 7 USB GPS adapter, and I can't get it to work with this build, I've tried CGPS in terminal and it just times out every time.

Also Conky doesn't display, when I type in Conky in terminal I get the following error;

conky: Syntax error (/home/pi/.conkyrc:1: '=' expected near 'yes') while reading config file.

reading around it looks like the conkyrc file isn't compatible with the new version of Conky but I'm probably wrong.

Any anyone offer any help?

km4ack, 2020-04-26:
Does the GPS have a green blinking light on it?

uksoundz, 2020-04-26:
Hi, it does, I've done some further tinkering, and I can get Conky to show up if I type in Conky into the terminal, however it disappears if I restart, so I always have to open the terminal.

I've also got the GPS to work, but I have to type the following into terminal every time I start the Pi
sudo killall gpsd
sudo gpsd /dev/ttyUSB0 -F /var/run/gpsd.sock

Anything that can be done to prevent me having to go to the terminal every time?

Thanks!

km4ack, 2020-04-26:
Do you have the pi set to autologin? Conky won't start correctly if you don't use auto login