km4ack/pi-build #61: How to put on CONKY the "MAIN" and "SUB" radiofrequency?

Klammeraffe2, 2020-04-28:
Hello, 
have just made the installion of V2 and enjoy it very much - thanks. One thing I am looking for - how can I add beside the Main-VFO(A)-radio freq (from FLrig) also the VFO-B Radio-freq into the conky-info-window? 
Found thatthe info is taken with  "get-freq" - but don't see, how I can maybe add the VFO-B radio-freq?

km4ack, 2020-04-28:
added to list of possible enhancements in the future. It will require modding the get-freq script or writing a new script to read the VFO-B using hamlib &/or FLRIG 73, de KM4ACK

Klammeraffe2, 2020-04-28:
Tnx

closed 2020-04-28 by Klammeraffe2

chrisBosse, 2020-04-28:
@km4ack, Thank you for giving us all so much!

@Klammeraffe2 Some feature details could get a little messy, (see end thoughts).

All that being said, here is a possible modification until the feature is added.

Use at your own risk.

Assuming default user `pi`.

## 1. Modify `get-freq` Script

Replace the contents of `/home/pi/bin/conky/get-freq` script with this one.

```bash
#!/bin/bash

#get freq of radio to display in conky
#optional parameter: VFOA VFOB
#20191217 km4ack
#20200428 modified

source "$HOME/patmenu/config"

MAIN () {

# If first argument, $1, is empty or unset, use 'VFOA' as its value.
VFO=${1:-'VFOA'}

FREQ=$($RIG -o f "$VFO")
FREQCH=$(echo "$FREQ" | grep error)

if [ -z "$FREQCH" ]
then
HZ_AS_MHZ "$FREQ"
else
MAIN
fi

}

HZ_AS_MHZ () {

#Inject decimal into number six places from right.
#First, and only, parameter expected to be a string of six or more digits

DP=.
SIX_DIGITS=\[0-9\]\[0-9\]\[0-9\]\[0-9\]\[0-9\]\[0-9\]
MHZ="${1%%$SIX_DIGITS}"
HZ="${1##$MHZ}"
printf "%s$DP%s\n" "$MHZ" "$HZ"

}

MAIN "$1"
```

## 2. Add `get-vfo` Script

This lets Conky know which VFO `hamlib` thinks is active.

Add this new script as `/home/pi/bin/conky/get-vfo`.

```bash
#!/bin/bash

#get VFO of radio to display in conky
#20200428

source "$HOME/patmenu/config"

MAIN () {

VFO=$($RIG v)
VFOCH=$(echo "$VFO" | grep error)

if [ -z "$VFOCH" ]
then
echo "$VFO"
else
MAIN
fi

}

MAIN
```

Make sure the new script is executable.
`chmod +x /home/pi/bin/conky/get-vfo`

## 3. Modify Conky Configuration

Next, modify your Conky configuration file replacing the `Radio Freq` with something like this:

```lua
${font Arial:bold:size=18}${color White}Radio Freq A ${if_match "VFOA" == "${execi 5 $HOME/bin/conky/get-vfo}"}*$endif\
${alignr}${color Yellow}${execi 5 $HOME/bin/conky/get-freq VFOA | cut -c1-7}
${font Arial:bold:size=18}${color White}Radio Freq B ${if_match "VFOB" == "${execi 5 $HOME/bin/conky/get-vfo}"}*$endif\
${alignr}${color Yellow}${execi 5 $HOME/bin/conky/get-freq VFOB | cut -c1-7}
```

Note: I am using `conky-large` as my base Conky configuration.

## Ending Thoughts

I do not know what would happen if the reported frequency is less than 6 digits.

I have only tested this on my setup: a Raspberry Pi 3B+ and a Kenwood TS-450SAT using `flrig`.

Other thoughts while writing this modification.

- Frequencies below 1MHz (6 digits) and above 1GHz.
- International decimal point character, `.` vs. `,` vs. others.
- What happens if there are more than two VFOs? If there is only one VFO?
- What happens if the VFO is something different, like Memory? (In `hamlib` VFO can be `VFOA`, `VFOB`, `VFOC`, `currVFO`, `VFO`, `MEM`, `Main`, `Sub`, `TX`, `RX`.)
- What happens if the VFO is changed on the rig, but not with `FlRig`? On my setup, `hamlib`/`flrig` does not report changes to VFO A/B when I change it on the radio.