km4ack/pi-build #65: Programs close when I try to use them

KN4AHN, 2020-05-13:
I have a Pi4 with the pi build v2.  FLrig has control of my rig, and in the other programs (fldigi, wsjtx) I have set to FLrig FLrig, but when I use any function that does anything with the CAT control, the program closes.  Any thoughts as to what it could be?

km4ack, 2020-05-13:
Sounds like an issue between the applications. Try posting the question in the FLDIGI forum. They will probably have an answer for you. https://groups.io/g/linuxham/topics Please report back what you find.

closed 2020-05-17 by km4ack

KN4AHN, 2020-05-21:
Jason,
Thank you for your help, sorry it took so long to get back to you, I don't get as much time to tinker with stuff as I would like.  The short version is I am a dummy and didn't have my settings right.  

The long version starts with me explaining that I don't have a clue with what I am doing.  I had been interested in getting a pi for awhile, but with no linux experience, I was hesitant.  I decided, what the heck, lets do it, and got one for christmas.  Found the linux downloads on websites (like for wsjt-x), and voila!, didn't work.  Yeah, voila is the wrong word, but that is what happened in my brain before i realized that what I had wanted, and what was happening were very different things.  So I started looking online for instructions, and at the time, none of them made sense.  I had not worked from a command terminal, and just couldn't get my head wrapped around the concept.  I tried looking up tutorials, but nothing I found made sense, and as a working dad, the time that I actually have to be involved with the hobby is extremely limited, so I don't have the time or energy to learn linux like I should.  After a couple frustrating weeks of just not understanding, I came across your tutorials, where you break it down step by step.  I started to understand!  After watching a few times, I broke out a pad of paper and started writing down where to click, and commands in a way that I could follow.  Before I started the installing, you posted the ultimate key to my success.  The build-a-pi script.  My mind was blown, its what I needed.  Followed your instructions to the letter and *boom*, it was flawless.  Got it connected to my rig, and it all worked (i think, didn't use all the programs, but happy they are there for when I want to try).  Then you came out with build-a-pi v2, which had even more options, to which I said 'I must'.  Smart enough to know I had something working, I saved up a little money and bought another microSD card, figuring I could set that up, and when that works, copy it onto my original card as my backup.  Followed your instructions on how to flash buster onto the card (my original came with the pi and was already set up), ran the new script, and it all worked!  Well, sort of.  After I set up FLrig (and when I clicked the 'Init.' button, the radio clicked and seemed to work), my programs would just close on me.  It was beyond me.  I contacted you here, and you recommended the other group.  They suggested I use the command prompt to open the programs (didn't know that was a thing) and it might throw an error message.  It did, and they responded that it seemed there was an error, to which I was told 
'That's a bug in the application.  it's probable they've tried to call free on something that's already been free'd.  Check you've got the current version check the project/distribution bug tracker...'
Now with my 'extensive' background experience with linux, I can honestly say that this did not help me.  I had been switching between  my SD cards, making sure all my settings had been identical, and they were, so I decided start with a clean slate.  Being the windows user I am, I know how to erase a memory card, right?  Wrong.  I stuck it in my wife's laptop (the pi is my only computer as well), deleted everything off of it, tried to flash Buster, only to find out there are only about 300+MB of memory on my 32GB card.  Almost lost my mind.  But I had my backup so I wasn't too concerned.  Then as I was trying to fix my error (keep in mind I only get short bits at a time to tinker so this took place over a few days) I realized I somehow had switched SD cards and my backup was showing 400MB of memory and was completely useless too.  That was my limit.  Too much frustration, and I hadn't moved back to square one, I had moved beyond that.  After I calmed down from my internal catastrophic breakdown I started researching again.  I learned that the reformatting tool on the laptop was not the one to use.  I downloaded the correct one, and started the process.  Fast forward a few hours, it worked, buster installed, build-a-pi script run, and Voila!  Back to the same problem as I started with, except I was a step behind because I no longer have a backup.  Again, frustration sets in.  I then research and write down for myself how to uninstall programs, and started with my most necessary programs to reinstall from source (FLrig, FLdigi, wsjt-x.  I am finally understanding a little of this linux stuff).  Something inside told me I was going to end with the same problem, so I decided to perform one last ditch, hail-mary effort. I opened FLrig, went to settings, went to my serial port drop down list (there are about 40-50 options it seems like).  It was set to /dev/ttyUSB0, which is what it was on the working setup (I should know, I wrote it down in like 4 spots during all this), and appeared to work fine on this one, changed it to /dev/serial/by-id/usb-Silicon_Labs_CP2105_Dual_USB_UART_BRIDGE_CONTROLLER_00ABFE6C-if00-port0.  As of right now, everything is working.  Not sure exactly how or why it does, but I just needed to share my roller coaster of a story with you, hopefully you had some chuckles along the way at my sanity.  It is easy for me to look back and laugh now that it was successful, and hopefully this story may help keep someone else from not giving up.  Thank you so much for all you have done to help those like me that don't have a clue still have fun.
George
KN4AHN