km4ack/pi-build #66: WSJTX not installing

km4ack, 2020-05-13:
not working correctly as of 20200513 syntax wrong when looking for latest package. returns 2 results and should only return 1

Rescue9, 2020-05-23:
Not necessarily a bug in the script, but more in the way the devs have added a RC to the link. A simple fix would be to strip the release candidate filename from the WSPKG. This will work "assuming" that the RC files are always listed after the General Availability files. 

echo $WSPKG | sed 's/[ ][^ ]*//'

A secondary option would be to include an if statement asking the user if they want to run the GA or the RC version, then stripping the file of either filename. Personally though, I'd recommend sticking with the GA version to keep the non-linux people safe should the RA version contain bugs.

km4ack, 2020-05-24:
Correct. I was referencing the update script. More of a note to self. If you would be interested in beta testing something, shoot me an email. mycall@arrl.net Please reference this post. 73, de KM4ACK

Rescue9, 2020-05-24:
Definitely in for beta testing. Since this codid thing I've gotten bored. Whatcha got? I'll setup my got properly to submit patches as well rather than just commenting.

AB (Reacue9 on GitHub)

PR #69 (2020-06-20): WSJTX not installing
Fixes #66

Remove instances of release candidate.
closed 2020-06-20 by km4ack