km4ack/pi-build #67: VOACAP - Can't select antennas

wb8erj, 2020-05-20:
When I click on Antennas to select antenna, it opens up the antennas folder but does not show the samples folder. I tried copying and antenna file from samples folder to the antennas folder and it still does not show up.
[image]

km4ack, 2020-07-11:
This could be related to updates in Buster or one of the dependencies that application relies on.

There are newer versions of both VOACAP and PythonProp that need to be updated in the BAP build. As of 11JULY2020, I haven't successfully installed the updates.

fhules2, 2020-07-11:
I am also not able to see the antennas.