km4ack/pi-build #78: change RTC to use config.txt

nfg42, 2020-06-23:
Raspbian has the option to support RTCs built in. 

You just have to add the line like this "dtoverlay=i2c-rtc,ds1307" to the /boot/config.txt file. 

[raspberry-pi-rtc](https://pimylifeup.com/raspberry-pi-rtc/)

km4ack, 2020-06-23:
Thank you for the heads up. I'll read up on it. 73, de KM4ACK

km4ack, 2020-10-14:
While modding the /boot/config.txt file will work, it requires a reboot which is avoided during the build so the user can answer a few simple questions and then walk away while the build happens. 

The RTC is an option to add during the update process. I have left it up to the user to enable I2C interface using the built in Pi tools. The script also gives the user a warning if this hasn't been done but they try to install the RTC.

closed 2020-10-14 by km4ack