km4ack/pi-build #81: Remove ‘pi’ user install requirement

JaySheepdog-N0SR, 2020-06-26:
Allow installation of BAP using the current logged in user, checking for and disallowing ‘root’.

‘pi’ is a default user, and current security practices dictate that these type of accounts be disabled (or secured and left unused) and individual user accounts be created instead.

Taubin, 2022-02-21:
I just wanted to add my two cents here, I'd love to be able to run this as another user. I am building a pi to toss in at my local ham club, and do not want the user exposed to the club users to be root for obvious reasons. I'd love to run this as another user, then remove them from the sudoers.

km4ack, 2022-08-26:
This was implemented in 3.2.4

closed 2022-08-26 by km4ack