km4ack/pi-build #82: Conky freezes when gps is plugged in

plowboy68, 2020-06-28:
All of the information on my conky display freezes shortly after the gps dongle is plugged in. System clock runs normally. Unplug gps, conky starts running normally.

Wirezfree, 2020-07-15:
Does it freeze when PI is booted with GPS already connected?

plowboy68, 2020-07-15:
It does not. I changed the “hotplug” setting in gpsd to true at some point and that was causing the problem when the Pi is already running. If the gps is plugged in at boot with hotplug set to false there is no problem.

Wirezfree, 2020-07-15:
> It does not. I changed the “hotplug” setting in gpsd to true at some point and that was causing the problem when the Pi is already running. If the gps is plugged in at boot with hotplug set to false there is no problem.

O.K... Will have to see if the other actions on GPS enhancements address this

km4ack, 2020-10-14:
If you are still having this issue, see [this video](https://youtu.be/YIIC-iT6H-s) for the fix.

closed 2020-10-14 by km4ack