km4ack/pi-build #83: Conky started incorrectly

pleasantone, 2020-07-01:
I'm creating a separate issue for this, which was mentioned in #60 

Conky is currently being started from user pi's crontab. That assumes autologin and all sorts of other badness (like locking things down to the pi user) and race conditions.

The proper way to start applications that should fire up upon login is using ~/.config/autostart.

My suggestion, take this snippet, put it in ~/.local/share/applications/ and then symlink that file into ~/.config/autostart. This means it will both show up in the menu (under hamradio) as well as autostart.

Snippet follows:
```
[Desktop Entry]
Name=Conky
Comment=Conky
GenericName=Conky Screen Background Monitor
Exec=conky
Type=Application
Encoding=UTF-8
Terminal=false
Categories=HamRadio
Keywords=Radio
```

km4ack, 2020-07-07:
slated for 3.0.3 release

closed 2020-08-17 by km4ack

kc917, 2020-09-20:
When I shut down or reboot the system, Conky freezes. Editing .conkyrc or running setconky restarts. Is there a delay for coney on restart?

km4ack, 2020-09-20:
Which BAP version are you running? There was a change recently to the way Conky is started at boot.

kc917, 2020-09-20:
version 3.0.3, sorry forgot to mention that

km4ack, 2020-09-20:
Ok. That version has the new way to start conky. Are you able to start from the ham radio menu after boot or just by running set-conky?

Also, do you have a conky.desktop file if you run 
ls .local/share/applications/

kc917, 2020-09-20:
I didn’t try starting it from the menu. will give it a try.  the conky.desktop file is in .local/share/applications/  conky starts on boot and reboot but is froze with nothing updating. it’s been years since i’ve fooled with linux. used to run a packet bbs here in northern WV. 
I’ll let you know what transpires with starting from menu.

kc917, 2020-09-20:
Ok, it starts from the menu after reboot and freeze. Not a major problem, just minor annoyance. Thanks for your time. Ken