km4ack/pi-build #85: WSJTX Auto Grid from GPS

blackwellj, 2020-07-03:
https://github.com/bmo/py-wsjtx/tree/master/samples @km4ack

_Originally posted by @blackwellj in https://github.com/km4ack/pi-build/issues/84#issuecomment-652693701_

Appologies, I cant seem to find how to reopen an Issue

km4ack, 2020-07-03:
I'll look into this. At first glance, it appears to be written for windows. Python is cross platform but seeing COM8 in the code leads me to believe it was for intended for Windows. I might be able to mod the code for GPSD but I don't typically code in python.

km4ack, 2021-08-06:
As no work has been done on the code referenced in the past 4 years, I am closing this issue. Line 19 of [his code](https://github.com/bmo/py-wsjtx/blob/master/samples/grid_from_gps.py) tells us that he hasn't implemented this for gpsd on Linux. Feel free to open a new issue ticket if the code is updated in the future.

closed 2021-08-06 by km4ack