km4ack/pi-build #86: GPS/GLONASS U-Blox7

Cbruckman2019, 2020-07-04:
I purchased the GPS/GLONASS U-Blox 7 from Amazon and it works intermittently with Conky.  Even after system reboot the dongle doesn't work. Again it works intermittently. I can come back in a hour and it is working...then an hour later it isn't.

km4ack, 2020-07-04:
Are you using the USB 3.0 ports for anything? When Conky says NO GPS, does the GPS have a lock? If you are uncertain, wait until Conky says NO GPS and then run "cgps" in a terminal window and verify the GPS actually has a lock.

Cbruckman2019, 2020-07-05:
I moved it to the 3.0 USB port and it started working. Conky was showing
the Gridsquare but when I ran the cgps command the gps lost signal.

km4ack, 2020-07-05:
thanks for the update

closed 2020-07-05 by km4ack