km4ack/pi-build #91: Fldigi and wsjtx don’t install

g0cer, 2020-07-07:
I installed my brand new pi4 with buster and ran PI build V3 on it 
I Selected everything I wanted but fldigi and wsjtx didn’t install I see it running against the fldigi build files but it end with telling me any errors and offers the reboot screen but on reboot none of the two are there either in menu or when trying to run in terminal.
I think I saw errors in python install? But not sure in the building terminal. 
Thanks Dave G0CER

km4ack, 2020-07-07:
Thanks for the report. I'll try a fresh build this morning to see what's going on.

73, de KM4ACK

km4ack, 2020-07-07:
I ran build a pi on a fresh copy of buster. Selected just the 2 apps you had issues with and both installed ok. You may have had an internet glitch during installation or the servers may have been down for maintenance. Try using the BAP update script to install these apps. From main pi menu choose Preferences->Build a Pi. If this fails, please post in the forums until we can confirm it is a bug. https://groups.io/g/KM4ACK-Pi/topics

73, de KM4ACK

closed 2020-07-07 by km4ack

g0cer, 2020-07-07:
Thanks will do when home from work
Dave g0cer