km4ack/pi-build #93: Feature Request: TQSL

blackwellj, 2020-07-09:
Would it be possible to add TQSL for logging to LOTW?

km4ack, 2020-07-09:
Thanks for the suggestion. I'll look into it and see what I might be able to figure out. 

73, de KM4ACK

lcgreenwald, 2020-08-02:
Hi Jason,
I installed TQSL with these commands after a successful Build-a-pi v3.0.3 beta install.

sudo apt install -y cmake g++ libexpat1-dev libssl-dev libdb++-dev libcurl4-openssl-dev libwxgtk3.0-dev
git clone https://git.code.sf.net/p/trustedqsl/tqsl trustedqsl-tqsl
cd ~/trustedqsl-tqsl
cmake .
make
sudo make install
sudo cp /home/pi/trustedqsl-tqsl/apps/tqsl.desktop /usr/share/applications/

km4ack, 2020-08-02:
Thanks @lcgreenwald On the map for 3.0.4 release

closed 2020-09-24 by km4ack