km4ack/pi-build #99: backup config script??

km4ack, 2020-07-15:
Possibly write a script that would backup/restore many of the users config files for various applications like js8call, fldigi, pat menu, etc?

Wirezfree, 2020-07-15:
**"Build A PI ~ BackUp & Restore"**

**Purpose:**
Allow the BackUp and Restore of "ALL" user and configuration data related to the "Build A PI" environment.
This can be copied to "Removable" media(USB Stick)

**Aim:**
Allow quick "Restore" of user and configuration data either after a "New Build" or "Install Corrupted"
This will facilitate a quick install or recovery.

Dave - G8LIY

SpudGunMan, 2022-10-20:
possible idea here, this is just a backup right now https://github.com/km4ack/pi-build/pull/476