km4ack/pi-build #101: Blanket rm statements will lead to user files being deleted

jangliss, 2020-07-16:
Lines 488 to 492 need to be looked at.  I download files to my Downloads folder, the last step here is to delete everything in that folder.  

Other cleanup stuff here might need to be weighed in on user preference, such as the "apt -y autoremove", I might have packages listed as unused, but still be using them, the package manager just might not know it (for example libraries I installed using apt and then manually built against).

km4ack, 2020-07-16:
Good catch. This was left over from version 1 & 2 which were only designed to be run on a fresh build. Labeling as a bug to move up the priority list.

km4ack, 2020-08-17:
Slated for 3.0.4 release