km4ack/pi-build #102: cqrlog 2.4

RAbagnale, 2020-07-16:
If you remember I commented one of your videos with a workaround to update cqrlog from version 2.3 to 2.4 to get eqsl running out of cqrlog (which is not possible with version 2.3). 
Maybe it would be a good idea to ad this workaround into the coding of build-a-pi so that it installs automatically.

griffeyt53, 2020-07-17:
RAbagnale - I don't believe cqrlog is being built from source and is being pulled from the repository.  Agree that 2.4 would be better.  The wsjt-x integration in the 2.3.0 version has issues regarding remote logging and is documented by the developers/supporters. Thanks Jason for your great work and effort.  Some mighty fine scripting!

RAbagnale, 2020-07-17:
griffeyt53,
You might be right. But there are just a very few commands to execute to upgrade to 2.4. I'm sure this could be implemented in the code. Well, let's see what Jason thinks about this.

km4ack, 2020-09-25:
@RAbagnale I can't seem to find the comment you left. Can you refresh my memory concerning the workaround please?

RAbagnale, 2020-09-25:
Hi Jason,
I sure can:

#First we get the dependencies sudo apt-get install lcl*2.0 lazarus*2.0 fp*3.0.4 fpc*3.0.4 libssl-dev git build-essential libmariadbclient-dev
 #Next we do a kludge so CQRLOG finds the db client libraries it needs sudo ln -s /usr/lib/arm-linux-gnueabihf/libmariadbclient.so /usr/lib/libmysqlclient.so
 #Finally, we do the build and cleanup git clone https://github.com/ok2cqr/cqrlog.git cd cqrlog make sudo make install cd .. rm -rf cqrlog*
And that's it. You're done. Typing cqrlog at the prompt will start the program.

73,
Rico

km4ack, 2021-01-11:
slated for 3.0.6 release.

closed 2021-02-17 by km4ack