km4ack/pi-build #103: Direwolf support for CM108 Audio Adapter

b33ngaming, 2020-07-19:
Hi Jason, I love the build a pi script, 

I would like to suggest adding the support for the CM108 adapters as described in section 5.2.4 of the [Direwolf User Guide](https://github.com/wb2osz/direwolf/blob/master/doc/User-Guide.pdf) to your build a pi script.  I personally have a [RA-35](http://www.masterscommunications.com/products/radio-adapter/ra35.html) USB audio adapter, and editing line 195 in direwolf.conf to PTT CM108 gives the error in the attached picture.

Thank you for taking the time, and this is a great thing you've put together!

[image]

b33ngaming, 2020-07-19:
I failed to mention that this isn't a blocker.  Installing the package and running Build a Pi again for Direwolf took care of it.  Everything is working as I hoped it would.

km4ack, 2020-07-20:
Just to make sure I understand correctly, you ran Build a Pi again and that solved the problem?

73, de KM4ACK

b33ngaming, 2020-07-22:
Yes sir, I installed libudev-dev with `sudo apt install libudev-dev` and then ran build a pi again, only selecting Direwolf.  Direwolf checks for the package and then configures it when compiling I guess.  :)

Thanks for your time!

closed 2020-07-22 by km4ack