km4ack/pi-build #128: XGPS or similar?

hicksdp, 2020-09-08:
In your video about diagnosing the GPS issue on Field Day, you showed XGPS as a means to see satellite status.  Can you add XGPS or something similar to the next build?  Having the Grid Square display in Conky is nice, but sometimes I'd like to see the GPS coordinates.  Thanks, N4DPH

km4ack, 2020-09-08:
If you install the GPS software with BAP, xgps is there for your use. Start it from the command line with "xgps"

hicksdp, 2020-09-08:
Thanks-  XGPS appears, but the fields are blank (even though Conky is showing my grid square).  Also, the window runs off the bottom of the screen and won't resize.

km4ack, 2020-09-08:
Interesting. Usually it's the other way around. Do you have data with "cgps"?

hicksdp, 2020-09-08:
No data with cgps either.

David Hicks (256) 565-4020

km4ack, 2020-09-08:
Run this command and post the results please
cat /dev/default/gpsd

hicksdp, 2020-09-08:
Sorry for the confusion. Another reboot solved the problem. Now a god and cgps are delivering data.

David Hicks (256) 565-4020

km4ack, 2020-09-08:
Thanks for the update.

closed 2020-09-08 by km4ack