km4ack/pi-build #131: HAMLIB showing needs update when none exist

km4ack, 2020-09-10:
Per this forum post - https://groups.io/g/KM4ACK-Pi/topic/76748159?p=Created,,,20,1,0,0

robvdveeke, 2020-09-10:
I found the problem 
when you remove the directory package.nw from the directory Gridtracker and do the installation than it works 
I moved package. nw to an other directory I do not know if I need it so can always put it back 
73,s PA3BGQ

km4ack, 2020-09-10:
I think you meant to post this to this issue https://github.com/km4ack/pi-build/issues/130

robvdveeke, 2020-09-10:
O dear you are correct my bad

km4ack, 2020-09-24:
The Build a Pi (BAP) script is currently hard coded to download HAMLIB v3.3 even though it checks for a newer version of HAMLIB.

The latest version of HAMLIB (4.0rc2) is currently a release candidate. Reading through some of the release notes indicates that the rig numbers have changed in this version to accommodate future growth of the HAMLIB code. This may have an negative impact on current users running BAP. Although most users use FLRIG (which didn't change), there will need to be a post/video made when the final 4.0 version is released. Due to these factors, I am choosing not to resolve this issue at this time and instead wait for the final release of 4.0. At that time, BAP may need to be updated to dynamically download the latest version or hard coded for the latest version. Currently I recommend hard coding each version after it is released and tested as HAMLIB is a foundational piece of code to BAP. If set to dynamically download the absolute latest version, this could leave users not interested in beta testing running a version of HAMLIB that isn't stable.

km4ack, 2021-01-08:
Is direwolf 1.6 compatible with Hamlib 4.0? Reference [this post](https://groups.io/g/direwolf/topic/79501429)

km4ack, 2021-01-27:
It seems there might be some additional issues that need to be investigate before baking hamlib 4.0 into BAP See [this post](https://groups.google.com/g/pat-users/c/_ClBgzzGeoo)

km4ack, 2021-01-29:
hamlib 4.0 is no longer a RC.

km4ack, 2021-02-17:
the fix will be included in 3.0.6

closed 2021-02-17 by km4ack