km4ack/pi-build #144: gpsd reconfigured my GPS making it unusable on Windows

hwdornbush, 2020-09-20:
I connected my BU-353S4 USB GPS to BAP and found that it works fine there. with YAAC  But when I moved it to Windows, I found that it would not work.  YAAC's author told me that you must start gpsd with the -b switch so that gpsd will not change the configuration of the GPS but will leave it working as a NEMA-0183 device which is compatible with Windows.  So far, I have not found a way to un-do the reconfiguration, so my GPS can not be used with Windows.

hwdornbush, 2020-09-20:
I found how to do the -b switch.  

$ sudo nano /etc/default/gpsd
and change the last line to:

GPSD_OPTIONS="-n -b"

km4ack, 2020-09-20:
Thanks for the report. Not running Windows, I haven't run into this. You are the first to report the problem. Should be easy enough to add the extra switch in the next version.

hwdornbush, 2020-09-21:
I loaded BAP from scratch and edited the config file after BAP completed
its work but before I rebooted the Pi.  That worked.  Gpsd did not alter
the configuration on my GPS.  I am able to move my GPS between the Pi and
Windows.

Thanks for doing this.  It will really help in the future.  For now, I am
training my students (from my radio club) on how to alter the config file
before gpsd takes over.  A lot of them are just learning Pi and Linux, and
most still use Windows regularly.  I will be able to show them how to set
up YAAC on both BAP and their Windows computers.

--Bill AA6BD

km4ack, 2020-09-21:
Glad you got it figured out. This will be resolved in the next beta version. 3.0.4 beta was just released so it will be the version after that. Out of curiosity, what did you do to remedy the GPS to work with Windows again?

hwdornbush, 2020-09-21:
I followed the instructions at
https://spotwise.com/2014/11/26/switch-bu-353-to-nmea-mode/

I had to stop gpsd and reboot in order to get the commands to work but then
they worked perfectly.

--Bill AA6BD

km4ack, 2020-09-22:
Thanks for the link. I have cross posted this to the [forum](https://groups.io/g/KM4ACK-Pi/topic/bu_353s4_usb_gps_on_won_t/77010874?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,77010874) as well.

km4ack, 2020-09-24:
The -b switch has been added to the gpsd file - https://github.com/km4ack/pi-scripts/blob/master/gpsd

closed 2020-09-24 by km4ack