km4ack/pi-build #146: JS8Call Utilities doesn't always get grid square

km4ack, 2020-09-23:
Solution submitted by Dave - N9QNZ 

#stop the gpsd service
sudo systemctl stop gpsd

#add multi-user wants
sudo ln -s /lib/systemd/system/gpsd.service /etc/systemd/system/multi-user.target.wants/

#start gpsd service
sudo systemctl start gpsd

km4ack, 2020-09-24:
added to GPS [install script](https://github.com/km4ack/pi-scripts/blob/master/gpsinstall). See [this commit](https://github.com/km4ack/pi-scripts/commit/484b7a0e85ca43dfec068c8849ed6f031fe65313)

km4ack, 2020-10-14:
slated for 3.0.5 release. Currently in dev branch.

closed 2020-10-14 by km4ack

nq0m, 2021-06-22:
Hi Jason!  Just a minor mention on this - rather than explicitly create a symlink, I would instead do "systemctl enable gpsd" - which actually does the exact same thing:

$ sudo systemctl enable gpsd.service
Synchronizing state of gpsd.service with SysV service script with /lib/systemd/systemd-sysv-install.
Executing: /lib/systemd/systemd-sysv-install enable gpsd
Created symlink /etc/systemd/system/multi-user.target.wants/gpsd.service → /lib/systemd/system/gpsd.service.

ls -l /etc/systemd/system/multi-user.target.wants/gpsd.service
lrwxrwxrwx 1 root root 32 Jun 21 22:21 /etc/systemd/system/multi-user.target.wants/gpsd.service -> /lib/systemd/system/gpsd.service

IMHO, it's always best practice to use the proper systemd commands to enable/start services, rather than do explicit symlinks.  I have to say judging from the work on this project (which is impressive), you must be a Linux IT guy in $DAYJOB - or you really taught yourself well - coming from another ham who's been a Linux sysadmin for the last 15 years!