km4ack/pi-build #151: build-a-pie

nz6e, 2020-09-27:
I posted the error message over on groups.io
Shortly after entering my callsign it errored out

Selecting previously unselected package bc.
(Reading database ... 93971 files and directories currently installed.)
Preparing to unpack .../archives/bc_1.07.1-2_armhf.deb ...
Unpacking bc (1.07.1-2) ...
Setting up bc (1.07.1-2) ...
Processing triggers for install-info (6.5.0.dfsg.1-4+b1) ...
Processing triggers for man-db (2.8.5-2) ...
/home/pi/pi-build/build-a-pi: line 100: /home/pi/pi-build/intro.txt: Permission denied
/home/pi/pi-build/build-a-pi: line 112: /home/pi/pi-build/intro.txt: No such file or directory
rm: cannot remove '/home/pi/pi-build/intro.txt': No such file or directory
/home/pi/pi-build/build-a-pi: line 169: /home/pi/pi-build/config: Permission denied
/home/pi/pi-build/build-a-pi: line 174: /home/pi/pi-build/base.txt: Permission denied

Looking at the script entry, it didn't copy and paste correctly, in the youtube the line is word wrapped and when doing a cut and paste it did some weird stuff.

Tried again with a fresh build and it couldn't find the library.

After a bad day and then this, I gave up.

Greg
NZ6E

km4ack, 2020-10-11:
Greg, did you get this resolved?

nz6e, 2020-10-12:
I’ve been able to build-a-pi successfully with the GPS Puck.

 

Having issues with the GPS hat. Way different configuration. Working on it.

The GPS is working with gpsmon, xgps, cgps -s

But I can’t figure out (yet) what I need to set in the refclock statement

The GPS is /dev/ttyAMA0 and it’s also pps0

And I can’t get it to communicate with Chrony.

 

I’ll figure it out!

 

Greg

km4ack, 2020-10-12:
Thanks for the update. Closing this issue but I am Interested in the project. Feel free to shoot me an email or post in the [forum](https://groups.io/g/KM4ACK-Pi/topics) once you get it figured out.

closed 2020-10-12 by km4ack