km4ack/pi-build #153: JTDX fails to install

km4ack, 2020-10-02:
Whether using the wget command or curl command, access denied is being returned. This is currently preventing JTDX from being installed during the build. The only way around this that I know is to hard code the download link into BAP.

km4ack, 2020-10-02:
as of 02OCT2020 the entire website shows access forbidden even when trying to navigate to https://www.jtdx.tech/en with a web browser. This may not be a BAP issue and instead may be a website issue.

km4ack, 2020-10-06:
As of 06OCT2020 the website appears to still be offline. Getting same access forbidden message in a web browser.

km4ack, 2020-10-14:
As of 14OCT2020 the website appears to still be offline. Getting same access forbidden message in a web browser.

km4ack, 2020-10-14:
jtdx removed from bap script & update script. Install code left in additional.function script for now. Will add back once JTDX website is restored

lcgreenwald, 2020-10-26:
As of 26Oct2020 the jtdx web site seems to be back up.  I successfully installed jtdx with BAP/master.

km4ack, 2020-10-27:
Interesting. I still get an access forbidden at https://www.jtdx.tech/en

W9CVO, 2020-11-03:
Checked as of 02NOV2020 and the site is working.  Is there any chance you've accidentally tripped some sort of DDOS protection on the site?

km4ack, 2020-11-03:
Not sure what is going on. On Firefox and Safari I still get the access forbidden. Using a VPN with another browser I can see the site. I never use Safari. Just opened it to check to see if I could access.

km4ack, 2020-11-03:
Just double checked this on a new pi build to verify it wasn't something with my browser. It appears that the webserver or author has blocked my ip from accessing the site. Until the author of JTDX resolves this issue, I will not be able to support JTDX within build a pi. New installation and updates of JTDX will not be available starting with 3.0.5. I'll be happy to add it back once this is resolved.

closed 2021-08-05 by km4ack