km4ack/pi-build #154: Hotpot enabled but not able to VNC into Pi

HITMAN19832006, 2020-10-04:
I installed Build a Pi 3.0.3 on a fresh SD card with Raspbian. Not sure if this is fixed with 3.0.4. but am unable to message KM4ACK directly.
I can force the Hotspot to start via the Hotspot tools. It's SSID can be recognized by devices such as my cellphone. I can connect to the Hotspot via wifi. 
However, I cannot VNC into the Pi despite being able to at home while the Pi is connected my home wifi.
I am told by VNC Connect that my VNC isn't open to the cloud for requests.
What are some items I should look at to resolve this? I don't see the double arrows associated with internet connection.
Any other ideas?

km4ack, 2020-10-04:
Are you using RealVNC? https://www.realvnc.com/en/connect/download/vnc/

What ip address are you trying to use to connect to the pi?

What device are you using to connect to the pi?

HITMAN19832006, 2020-10-04:
I am using RealVNC and Raspberry Pi 4 IP is 10.0.228

km4ack, 2020-10-04:
When you connect to the pi's hotspot, the pi has an ip address of 10.10.10.10 Try connecting to that ip addr instead.

HITMAN19832006, 2020-10-04:
Will test later and will reply with results

HITMAN19832006, 2020-10-05:
I was able to connect using that ip on my cellphone. Tried to do the same using a RealVNC apk on my kindle but got locked out.
Is the login the SSID you see in HTTools to VNC?

km4ack, 2020-10-05:
No. Your login is your pi username and password. 

What you see in hotspot tools is what you use in your wifi setting to connect to the hotspot.

HITMAN19832006, 2020-10-05:
That's probably the issue on my end. Thank you so much for clarifying. I'll try it again tomorrow and let you know if there are any problems

km4ack, 2020-10-05:
No problem. I am going to close this issue ticket because I don't think this is an issue with the BAP code. If you need more help, please post the question on the [forum ](https://groups.io/g/KM4ACK-Pi/topics) as many more people follow the forum and are willing to help get you up and running. 73, de KM4ACK

closed 2020-10-05 by km4ack