km4ack/pi-build #157: Pat Menu tries to launch browser at address 127.0.0.1_/

Darrrren, 2020-10-08:
When I click Start Ardop Modem in Pat Menu, everything launches and works as expected except the automatic browser launch. The browser window does launch correctly but it try's to connect to http://127.0.0.1_/ instead of http://127.0.0.1:5000/ui.

I've searched the issues and can't find anything so I'm pretty sure it's something I've done.  Any ideas anyone?

73's
Darren

km4ack, 2020-10-08:
Set the port number in your Pat Menu settings and try again.

Darrrren, 2020-10-09:
Thanks Jason. Which option is it I need to change?  Here are my settings:
```
Callsign:          SA6DAZ
MAP:               World
Rig Control:       Yes
ARDOP Command:     /home/pi/ardop/./piardopc 8515 plughw:2,0 plughw:2,0
ARDOP GUI Command: /home/pi/ardop/./piARDOP_GUI
Direwold Command:  direwolf -p
KISS Comnd:        sudo /usr/sbin/kissattach /tmp/kisstnc
AX25 Port:         wl2k
Rig Control Comnd: /usr/local/bin/rigctl -m 4
HF Mode:           USB
2M Mode:           FM
PAT Port:          8080
Log:               /home/pi/Documents/mylog.txt
AmRRON:            no
```
I'm in the wrong place right? :-D

Cheers

closed 2020-10-09 by km4ack