km4ack/pi-build #159: No Gridsquare or time update with connected GPS in Conky

b33ngaming, 2020-10-12:
I did a fresh install of Raspbian OS, and Build-a-Pi and conky no longer updates the gridsquare after GPS is locked.  I tried disconnecting from my home internet and powering off the pi for a little while.  When it boots, the time is off as to be expected, but after I get my GPS lock (LED on the dongle) the time doesn't update like it did.

I checked my Build-a-Pi version today, it is at 3.04, and I typed conky into the terminal, and I get the error shown in the screenshot, could be the issue, but I am not that versed in the workings of this software.  I used gpspipe -w in the terminal, and I can see my lat/long output, and elevation, so I believe the data is being streamed to the Pi.

[image]

km4ack, 2020-10-12:
If you run "cgps" do you have GPS data in the app?

b33ngaming, 2020-10-12:
Yes sir, as well as xgps, that shows ~10 GPS satellites.

[image]

b33ngaming, 2020-10-12:
Not sure if this is a helpful observation, but, usually in the older version of build a pi, I was able to plug the GPS is while the Pi was already booted, and it would work.  This time, I had the Pi off, and didn't have the GPS plugged in when I booted.  The green LED was blinking showing connection, but no data in xgps/cgps.  I had to reboot the Pi, with the GPS Dongle plugged in, and that allowed the data to stream finally.

km4ack, 2020-10-12:
It is designed so that the GPS has to be plugged in at boot. The GPS is can't be hot swapped. 

It looks like one of the ruby gems didn't install during the build. Install it manually with 
`sudo gem install gpsd_client`

See if conky will give you the gridsquare. You might have to reboot after installing.

b33ngaming, 2020-10-13:
I did another full install, and a fresh build a pi, no dice still.  I installed gpsd_client, didn't work after the reboot.  I had to install maidenhead, that was the next error after entering `conky` in the terminal.

I have a grid square now, I will try and see if the time updates with GPS tomorrow morning.

b33ngaming, 2020-10-13:
gpsd_client and maidenhead installs fixed my issue.  Just tested the update of system time, and it worked once the GPS locks.

Thanks for your help.

km4ack, 2020-10-13:
Glad that fixed it.

closed 2020-10-13 by km4ack