km4ack/pi-build #161: Wifi SSID with spaces not displayed tight

BiggerSnake, 2020-10-19:
I have an SSID like this:

The Smith’s Network

It displays near the start of running Build-a-Pi as

The

BiggerSnake, 2020-11-24:
The following should fix this problem. I tested it here with my SSID that has blanks in it. The idea came from a StackOverflow.com thread on trimming spaces in awk.
I found two different places with similar code: one in "build-a-pi" and another in "update" that both probably need this code

LIST=$(sudo iw dev "wlan0" scan ap-force | grep "SSID:" | awk 'BEGIN{FS=":"; OFS=" "} {gsub(/^[ \t]+|[ \t]+$/, "",$2); print $2}')
LIST=$(echo "$LIST" | tr '\n' '|')
echo $LIST

Jim Record - BiggerSnake
AD0YO

kuperman, 2020-11-30:
I had a similar issue and created a fix for the "build-a-pi" script (see below). Haven't had a chance to modify "update"

`LIST=$(sudo iw dev "wlan0" scan ap-force | sed -ne 's/^.*SSID: \(.*\)/\1/p')`

BiggerSnake, 2020-11-30:
This trims leading and trailing blanks from SSIDs.

LIST=$(sudo iw dev "wlan0" scan ap-force | grep "SSID:" | awk
'BEGIN{FS=":"; OFS=" "} {gsub(/^[ \t]+|[ \t]+$/, "",$2); print $2}')
LIST=$(echo "$LIST" | tr '\n' '|')

Jim
AD0YO

joshuajayg, 2020-12-18:
Same as this issue I opened in July: https://github.com/km4ack/pi-scripts/issues/22

km4ack, 2021-01-11:
This should be resolved with pull #182 which has been merged to the dev branch and is slated for release 3.0.6

PR #184 (2021-01-09): Add support for Home SSID with spaces in name
Fixes #161 

- grabs non-blank SSID entries
- sorts them
- removes duplicates
- joins them with '|' to use in the dropdown

I've got some ideas on how to get the currently used SSID to the top of the output so that it would be the default choice, but I'll need to work on that a bit and if I get it working will submit as a separate PR.

Benjamin - AA7BK
closed 2021-01-11 by km4ack