km4ack/pi-build #164: GPS Not Working

andjones132, 2020-10-29:
km4ack.

Hi. Thank you for pulling together a fantastic tool for the ham radio community. I have installed the pi build on my Raspberry 4 and everything seems to be working. That is except for my GPS. In the attached txt file you can see that the U-Blox GPS dongle has been recognized ad appears to be on port /dev/ttyACM0. If I run cat /dev/ttyACM0 on this port I get GPS data from the U-Blox dongle. However, cgps or xgsp both show no data. 

I checked the /etc/default/gpsd file and everything appears to be correct. 

```
pi@pibuild:/etc/default $ cat gpsd
#The -b switch was added per https://github.com/km4ack/pi-build/issues/144
#km4ack 24SEPT2020

# Default settings for the gpsd init script and the hotplug wrapper.

# Start the gpsd daemon automatically at boot time
START_DAEMON="true"

# Use USB hotplugging to add new USB devices automatically to the daemon
USBAUTO="false"

# Devices gpsd should collect to at boot time.
# They need to be read/writeable, either by user gpsd or the group dialout.
DEVICES="/dev/ttyACM0"

# Other options you want to pass to gpsd
GPSD_OPTIONS="-n -b"
```
Any ideas what I can try to get this working?

This is the dongle that I am using: https://smile.amazon.com/gp/product/B01MTU9KTF/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

km4ack, 2020-10-29:
Follow this video and it should get it going. https://youtu.be/YIIC-iT6H-s

andjones132, 2020-10-29:
Thank you, Sir. Your Awesome. That fixed it. Closing the issue and will use KM4ACK@groups.io in the future.

73,
Andrew/K9ACJ

closed 2020-10-29 by andjones132