km4ack/pi-build #167: Make -j 4 to use all cores

km4ack, 2020-10-31:
Hi, Jason!

I would like to make a proposal for pi-build - when calling "make"
command, to add "-j 4". That way the compiling will use all cores and
the build time shall be reduced a tot.

-- 
73 Daff, LZ1DAF

km4ack, 2020-10-31:
Recommended in this [post](https://groups.io/g/KM4ACK-Pi/topic/build_a_pi_3_0_5_beta_release/77939583?p=,,,20,0,0,0::recentpostdate%2Fsticky,,,20,2,0,77939583).

Jehreg, 2020-11-10:
Be careful, as make -j <num> introduces parallel building and assumes that the code being built has been properly "modularized".  I have been running Gentoo for years (a distro where everything is compiled before being installed) and there are a number of packages that have allergic reactions to the global "make -j <num>" setting I use.  Parallel building really helps with the kernel compilation, but is usually not that interesting for small packages, as the linker step cannot be parallelized anyways, and that's usually the step that can take the longest because of the Pi's slow block device speeds.

In other words, I don't recommend making this a "global setting", but can help certain packages.