km4ack/pi-build #176: Conky disappears

K0TPC, 2020-11-22:
I ran BUild-a-Pi for a Raspberry4 and everything worked fine.  Tried to follow the instructions for Conky to change to F instead of C.  When Conky restarted, it disappeared. From some postings maybe it slipped off to the bottom or side of the screen?  How do I get it back?

km4ack, 2020-11-22:
Try running conky prefs again and set the size to medium

K0TPC, 2020-11-22:
Still no luck.  I tried all three sizes and get the pop up saying it needs
to restart, I agree, then nothing.

Thanks,
Tim

K0TPC, 2020-11-22:
Something else really weird just happened.  I just tried to run Build-A-Pi
updates, nothing came up, I rebooted and while the pi booted up, it booted
to a grey screen and in preferences, both build-a-pi and Conky-Prefs icon
is the paper airplane.

I am completely new to Pi (and Linux for that matter), and hate to be a
bother but anything you could tell me would be greatly appreciated.

Tim
K0TPC

K0TPC, 2020-11-22:
And when I try to run preferences again and update I get an error that says Invalid desktop entry file: '/user/share/applications/build-a-pi.desktop'

I have no idea what I've done...

kj7dmv, 2020-12-14:
I have installed latest BAP over the 12/2/2020 Buster and I see no conky. I have tried all conky sizes. Haven't rebooted after each one....

km4ack, 2020-12-14:
This [video](https://youtu.be/YIIC-iT6H-s) should help if Conky isn't starting.

KO4AXV, 2020-12-15:
> And when I try to run preferences again and update I get an error that says Invalid desktop entry file: '/user/share/applications/build-a-pi.desktop'
> 
> I have no idea what I've done...

I have the same issue but I can still see conky, all I did was change the screen resolution and now it doesn’t work! 

Somebody help, please

km4ack, 2020-12-15:
KO4AXV,

Please post in the forum and let's see if we can figure it out https://groups.io/g/KM4ACK-Pi/topics

73, de KM4ACK

kj7dmv, 2020-12-18:
The conky GPS video worked for me. I would post in the forum but i'm not
finding it when I search

>> Jim Weisgram KJ7DMV

km4ack, 2021-08-05:
The initial issue was resolved in 3.0.8

closed 2021-08-05 by km4ack