km4ack/pi-build #179: Build-A-PI update does not work

jerseyj, 2020-11-25:
The splash screen pops up with text that ends with "Ready when you are" but there are no buttons on the screen. A cmd window is open in the background that is blank. Hitting the enter key closes the splash screen and cmd window. Running from a cmd window closes the splash screen. It appears that no actual actions are taking place. There is no indication if there are no updates to be made or anything else.

[image]

km4ack, 2020-11-25:
Looks like it is running off of the bottom of teh screen. Try moving the window up or try a higher resolution and you will see the buttons.

closed 2020-11-25 by km4ack

jerseyj, 2020-11-25:
Thank you, duh! I should have figured that one out easily!

I did have another quesiton…when I ran the update in the command window I saw it checking a number of applications, then in the pop up window I was presented with choices of things that I could check off to install/update. Is that supposed to happen ? I thought it would just check the items installed for updates ?

Thanks again for your help,
Jerry
KB2GCG
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
"Those who would give up essential liberty to purchase a little temporary safety, deserve neither liberty nor safety." - Ben Franklin

It's also true that those who would give up privacy for security are likely to end up with neither. 
It's not the years in your life that count, but the life in your years"

"Change is Inevitable, Growth is Optional"

"Without continual growth and progress, such words as improvement,

 achievement and success have no meaning." - Benjamin Franklin
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

km4ack, 2020-11-25:
It gives you the option of what you want to update. If for some reason you don't want the latest version, don't put a check by that app. This also gives you the option of installing an app that you may have passed on during the initial build.

jerseyj, 2020-11-25:
Thanks, I’ll put in an enhancement request to have an “update only what I have installed” option.

Jerry
KB2GCG
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++
"Those who would give up essential liberty to purchase a little temporary safety, deserve neither liberty nor safety." - Ben Franklin

It's also true that those who would give up privacy for security are likely to end up with neither. 
It's not the years in your life that count, but the life in your years"

"Change is Inevitable, Growth is Optional"

"Without continual growth and progress, such words as improvement,

 achievement and success have no meaning." - Benjamin Franklin
++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++++

jerseyj, 2020-11-30:
Thank you, that solved the issue.

Jerry

- "Defeat lasts one day, giving up lasts a lifetime."